DuPPadotnet/ArduinoDuPPaLib #1: The Name RESET creates a name collision with a declaration for the STM32F1 series

mitag, 2019-03-19:
I cannot compile the library for STM32F1 series. A declaration of the GCONF_PARAMETER "RESET = 0x80". collides with a declaration name "RESET" which is necessary for the STM32F1 series. Therefore I made a fork and renamed the declaration to ENCODER_RESET, then I can compile for STM32F1.
[i2cEncoderLibV2 ENCODER_RESET.txt](https://github.com/Fattoresaimon/ArduinoDuPPaLib/files/2985391/i2cEncoderLibV2.ENCODER_RESET.txt)

Fattoresaimon, 2019-03-19:
Hello,
Thank you for pointing out this issue, i will fix ASAP!

mitag, 2019-03-20:
Hello,
Thank you for the great project and the open source code!  I have compiled the example on a Nucleo-64 STM32F102RBT just to make sure it's not a peculiarity of the BluePill. But with the Nucleo board it's the same. This reminds me, in this example I had to change RESET to ENCODER_RESET.
[i2cEncoderLibV2 ENCODER_RESET_Nucleo-64.txt](https://github.com/Fattoresaimon/ArduinoDuPPaLib/files/2987051/i2cEncoderLibV2.ENCODER_RESET_Nucleo-64.txt)

Fattoresaimon, 2019-03-25:
Hello,
I have a new update of the library that solve the problem.
Please have a look: https://github.com/Fattoresaimon/ArduinoDuPPaLib/releases/tag/v1.2.0
Thank you!

mitag, 2019-03-25:
Hello,
I just downloaded your update and recompiled the example and now the compiler swallows everything. :)
[i2cEncoderLibV2  20190325.txt](https://github.com/Fattoresaimon/ArduinoDuPPaLib/files/3004585/i2cEncoderLibV2.20190325.txt)
 
Thank you!

closed 2019-03-25 by mitag