DuPPadotnet/ArduinoDuPPaLib #11: LEDRing issue with ESP32

skibum-za, 2020-06-22:
Just tried to compile up the demo for the LEDRing on my ESP32. It fails with :- 

`Arduino: 1.8.12 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"`
`In file included from C:\Users\ajm1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal.h:53:0,`
`                 from C:\Users\ajm1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:35,`
`                 from sketch\Demo.ino.cpp:1:`
`C:\Users\ajm1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal-gpio.h:35:27: error: expected unqualified-id before numeric constant`
` #define PULLUP            0x04`
`                           ^`
``
`C:\Users\ajm1\Documents\Arduino\libraries\ArduinoDuPPaLib-master\src/LEDRing.h:56:10: note: in expansion of macro 'PULLUP'`
`     void PULLUP(uint8_t pull);`
`          ^`
``
`C:\Users\ajm1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal-gpio.h:37:27: error: expected unqualified-id before numeric constant`
` #define PULLDOWN          0x08`
`                           ^`
``
`C:\Users\ajm1\Documents\Arduino\libraries\ArduinoDuPPaLib-master\src/LEDRing.h:57:10: note: in expansion of macro 'PULLDOWN'`
`     void PULLDOWN(uint8_t pull);`
`          ^`
``
`C:\Users\ajm1\AppData\Local\Temp\arduino_modified_sketch_714030\Demo.ino: In function 'void setup()':`
``
`C:\Users\ajm1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal-gpio.h:35:27: error: expected unqualified-id before numeric constant`
` #define PULLUP            0x04`
`                           ^`
``
`C:\Users\ajm1\AppData\Local\Temp\arduino_modified_sketch_714030\Demo.ino:36:11: note: in expansion of macro 'PULLUP'`
`   LEDRing.PULLUP(0b111); // 32k pull-up resistor for avoid ghost effect`
`           ^`
``
`C:\Users\ajm1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal-gpio.h:37:27: error: expected unqualified-id before numeric constant`
``
` #define PULLDOWN          0x08`
`                           ^`
``
`C:\Users\ajm1\AppData\Local\Temp\arduino_modified_sketch_714030\Demo.ino:37:11: note: in expansion of macro 'PULLDOWN'`
`   LEDRing.PULLDOWN(0b111); // 32k pull-down resistor for avoid ghost effect`
`           ^`
``
`exit status 1`
`Error compiling for board ESP32 Dev Module.`

It seenms that both PULLUP and PULLDOWN are used as part of the ESP32 core functionality.

I changed the LEDRing.h and LEDRing.cpp to use LRPULLDOWN/LRPULLUP and changed the demo code to then use those and it compiled and worked perfectly. I'm not sure that's a good fix however. But works for now.

Fattoresaimon, 2020-11-06:
Thank you for finding the issue!
I will work on it

emielskey, 2020-11-09:
Same here! thanks!!!

In file included from /Users/emielskey/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal.h:53:0,
                 from /Users/emielskey/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Arduino.h:35,
                 from /var/folders/r4/35x6nbbx2nj4n5bkdy4ghlcm0000gn/T/arduino_build_494568/sketch/Sine_wave_Generator_6.ino.cpp:1:
/Users/emielskey/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-gpio.h:35:27: error: expected unqualified-id before numeric constant
 #define PULLUP            0x04
                           ^
/Users/emielskey/Google Drive/Arduino/libraries/ArduinoDuPPaLib/src/LEDRing.h:56:10: note: in expansion of macro 'PULLUP'
     void PULLUP(uint8_t pull);
          ^
/Users/emielskey/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-gpio.h:37:27: error: expected unqualified-id before numeric constant
 #define PULLDOWN          0x08
                           ^
/Users/emielskey/Google Drive/Arduino/libraries/ArduinoDuPPaLib/src/LEDRing.h:57:10: note: in expansion of macro 'PULLDOWN'
     void PULLDOWN(uint8_t pull);
          ^

Fattoresaimon, 2020-11-14:
Hello,
I have soled the issue, please check!
I have also added the library explanation
Thank you!