DuPPadotnet/ArduinoDuPPaLib #13: Compiler warning in i2cEncoderMiniLib::autoconfigInterrupt

hb020, 2020-11-04:
Hi, 
when compiling (with platformio), I get this:

`..../Libraries/ArduinoDuPPaLib/src/i2cEncoderMiniLib.cpp:168:17: error: 'reg' may be used uninitialized in this function [-Werror=maybe-uninitialized]`

See
https://github.com/Fattoresaimon/ArduinoDuPPaLib/blob/e1c0c6487853e436f7977fe20d9226f1ee51063d/src/i2cEncoderMiniLib.cpp#L165

You probably want to add "= 0" to it, making
`  uint8_t reg = 0;`

gvansickle, 2020-11-04:
Yeah, it's early in my morning here but that definitely looks like a real defect: all control flows through that function use that local var uninitialized.  And it does look like @hb020 's "uint8_t reg =0;' is the correct fix.

Fattoresaimon, 2020-11-06:
Thank you,
I have added the initialization

closed 2020-11-06 by Fattoresaimon