DuPPadotnet/ArduinoDuPPaLib #16: Compile errors for teensy_RGB_encoder example

jrsphoto, 2020-11-29:
Trying to test some v2.0 encoder boards I have but I'm getting the following compile error:

`Arduino: 1.8.12 (Windows 10), TD: 1.53, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz (overclock), Faster, US English"

C:\Users\jrsph\Documents\Arduino\libraries\ArduinoDuPPaLib-master\src\LEDRing.cpp: In member function 'void LEDRing::LEDRing_ClearAll()':

C:\Users\jrsph\Documents\Arduino\libraries\ArduinoDuPPaLib-master\src\LEDRing.cpp:96:12: error: 'PWM_MODE' was not declared in this scope

   PWM_MODE();

            ^

Error compiling for board Teensy 3.2 / 3.1.`

What am I missing?

Fattoresaimon, 2020-11-29:
Hello,

I have tried but my compiled without problem.
Looks like the problem is in the library of the RGB LED Ring, that i have recently updated. 
Please update the library.

Thank you!

jrsphoto, 2020-12-01:
Yep, I was using an older version.  Updated and that fixed the problem.

Thank you