DuPPadotnet/ArduinoDuPPaLib #17: compile errors Implementing your library

jrsphoto, 2020-12-07:
Hey Saimon, I have two different programs using a Teensy 4.1, Arduino 1.8.12 IDE, and your latest, i2cEncoderLibV2.h.  One program compiles without problems.  The other program is giving me errors with my functions that handle  .onChange, .onButtonRelease, onMinMax, and .onFadeProcess.

I configure the events:

```
        I2C_Encoder[i2c_enc_cnt].onChange = encoder_rotated;
        I2C_Encoder[i2c_enc_cnt].onButtonRelease = encoder_click;
        I2C_Encoder[i2c_enc_cnt].onButtonDoublePush = encoder_doubleclick;
        I2C_Encoder[i2c_enc_cnt].onMinMax = encoder_thresholds;
        I2C_Encoder[i2c_enc_cnt].onFadeProcess = encoder_fade;
```
and for example, the encoder_rotated function looks like this:
```

void encoder_rotated(i2cEncoderLibV2* obj) {
  switch (obj->id) {
    case 0:
      // Volume encoder
      //  codec.lineOutLevel(obj->readCounterInt()); // Set Line-out voltage level
      if (obj->readStatus(i2cEncoderLibV2::RINC)) {
        Serial.print("Volume Up ");
      }
      else
        Serial.print("Volume Down ");
        Serial.println(obj->readCounterInt());
      obj->writeRGBCode(0x00FF00);
      break;
    case 1:
      // Gain encoder
      if (obj->readStatus(i2cEncoderLibV2::RINC))
        Serial.print("Gain Up ");
      else
        Serial.print("Gain Down ");
        Serial.println(obj->readCounterInt());
      obj->writeRGBCode(0x00FF00);
      break;
    case 2:
      //Filter encoder
      if (obj->readStatus(i2cEncoderLibV2::RINC))
        Serial.print("Flt Increment ");
      else
        Serial.print("Flt Decrement ");
        Serial.println(obj->readCounterInt());
      obj->writeRGBCode(0x00FF00);
      break;
  }
```
I'm getting a compile errors like the following, but for all the events, not just onChange :

```
ra8875:279: error: variable or field 'enc_rotate' declared void
 void enc_rotate(i2cEncoderLibV2* obj) {
                 ^
ra8875:279: error: 'i2cEncoderLibV2' was not declared in this scope
ra8875:279: error: 'obj' was not declared in this scope
 void enc_rotate(i2cEncoderLibV2* obj) {
                                  ^
variable or field 'enc_rotate' declared void
```
I've been trying to figure this out for a day now and it might have something to do with the Arduino IDE and auto-prototyping?

I've posted the code for both programs on a gist:  https://gist.github.com/jrsphoto/f257f607b948ddf7aad47ef6463ba077

The file backpack_sdr_jrs_ili9341_audiosdr_i2c_encoders.ino will compile and function;
The file ra8875_spectrum.ino  is the one that is giving me the errors.

If you can help, I would be grateful.

jrsphoto, 2020-12-11:
Any thoughts on this?

Fattoresaimon, 2020-12-11:
hello,
Where is declared the function "enc_rotate" ?
There is only "encoder_rotated"
I think you mixed the two function somewhere.....