DuPPadotnet/ArduinoDuPPaLib #18: ESP32 and mini I2C encoder

Mesihas, 2020-12-14:
I saw this project in different places and also Arduino forum some ember recommend it. This is not an issue I am trying to know if purchase this product.

In my project I have a rotary encoder connected to my ESP32 and its very unstable. eg: turn the encoder four steps to right and  the result could be:
1
2
3
4
3
4
5
6
7
7
8
9
8
7
9
10

other issues: if I move the shaft half way ( like not complete the step ) I will get values that are not correct. It seems to be a problem with ESP32 and ecoders.

My question here is: This I2C encoder fixes that problems with ESP32? of course I could take advantage of the I2C and free pins but is not the most important.

dgoadby, 2020-12-14:
I would need to know more about your encoder and the code you are using before commenting on your results on ESP32. However, I do use a lot of the I2CEncoders and they are very good. You don't have to worry about debouncing or missed steps you just get the current value using I2C. There is no problem with the "half way" situation that you describe either.

Fattoresaimon, 2020-12-14:
Well @dgoadby already answered :)
Thank you!

rowifi, 2020-12-23:
In ESP32 isn't there a requirement to use attachInterrupt() ?? Is this library suitable for ESP32 as it stands because I couldn't get the interrupt working using the example with callbacks .. Or did I miss something?

Fattoresaimon, 2020-12-23:
Hello,
Please check the correct pin!
In the example the interrupt is polled because with Arduino if i use a real GPIO interrupt the TWI library hang.
But with the ESP32 you can use the real one.
In the example WEMOS_OLED_Menu_Callback_EEPROM i have used the GPIO interrupt.