DuPPadotnet/ArduinoDuPPaLib #19: Encoder absolute position

Gandrasg, 2020-12-17:
Hi! I would like to use an encoder for multiple functions (MIDI CC controller) using a menu system I would tell the encoder which one to change.  I want to be able to use the controller on IOS too so the -1 / + 1 signal of the encoder is not enough for me (relative).  I would need the absolute position of the encoder which I would also need to store due to bank changes.  I guess your encodered is capable of this?

Fattoresaimon, 2020-12-17:
Hello,
Yes of course, the I2C encoder can manage 32bit numbers and the V2.1 also float number.
You can check the datasheet!

Thank you,
Simone

Gandrasg, 2020-12-17:
thanks for your answer
I think I found the answer to my question in the description on the 9th page. :)

Gandrasg, 2020-12-17:
i so happy

Fattoresaimon, 2020-12-17:
Good.
If you are also interested on the RGB Encoder, i will be back in stock next week!
Thank you

Gandrasg, 2020-12-17:
Luckily I have 4 in my drawer :)
When I run the examples I get this error I'm using Teensy 3.6. What am I spoiling?
`'PWM_MODE' was not declared in this scope
   PWM_MODE();
            ^
Error compiling for board Teensy 3.6.
`

Fattoresaimon, 2020-12-17:
mmm... please update the library
It's similar to this issue: https://github.com/Fattoresaimon/ArduinoDuPPaLib/issues/16

Gandrasg, 2020-12-17:
I just downloaded the directory and installed it. i have v2.1 encoder

Gandrasg, 2020-12-17:
I remember when I bought and tried them the example codes worked well. (Teensy rgb encoder)

Gandrasg, 2020-12-17:
Sorry,  OFFTOPIC your RGB ledring is superb - love it. 
Maybe you have data on what is the lowest current consumption when there is no color mixing and low brightness (but still usable)?

Fattoresaimon, 2020-12-17:
Thank you!
Please update the library now 😄

Gandrasg, 2020-12-18:
OK its work fine! :) thanks