DuPPadotnet/ArduinoDuPPaLib #20: EEPROM usage

Gandrasg, 2020-12-18:
Hi Saimon!

If I want to control 4 virtual encoders with 1 encoder, can I use this eeprom to store and recall the positions of the virtual encoders?
Is it possible to make additional divisions in it by shifting?

Fattoresaimon, 2020-12-19:
Hello,
Yes, but it's not automatic....
You have to write the EEPROM  every time.
The EEPROM Is freely accessible, you can store every data that you want. 
Remember that the EEPROM Is guaranteed for at least 100k cycle of writing. If you don't pay attention you can reach that limit easily...

Gandrasg, 2020-12-19:
Thanks. In the meantime, I realized that unnecessary use of EEPROM. An array will also be good. :)

closed 2020-12-21 by Gandrasg