DuPPadotnet/ArduinoDuPPaLib #21: Multiple encoders

Gandrasg, 2020-12-21:
Hi!
Can you help? I don't fully understand how to assign tasks by ID for multiple encoders.
If I use multiple encoders, how do I request information for a particular encoder? How can I associate behavior with an ID. For example: If the encoder ID is X, do you do it, and if the encoder ID is Y, do you do it?
```
void encoder_rotated(i2cEncoderLibV2* obj) {
  if (obj->readStatus(i2cEncoderLibV2::RINC))
    Serial.print("Increment ");
  else
    Serial.print("Decrement ");
  Serial.print(obj->id);
  Serial.print(": ");
  Serial.println(obj->readCounterInt());
  "if the ID = 0" writeRGBCode(0x0000FF);
  "if the ID = 1" writeRGBCode(0x00FF00);
}
```
Thanks in advance.

Gandrasg, 2020-12-21:
If I understand correctly, is obj -> "function" what points to and identifies the encoder ID in a given function? How can I reference the ID of this function in a for loop? (Encoder.id == 1).
I'm not a programmer, I have minimal knowledge of c ++, I'm just trying to understand the code.
Sorry for the silly questions.

Fattoresaimon, 2020-12-21:
Hello,
the "i2cEncoderLibV2* obj" is just a pointer to a class. And you can access to the member to the class with the operator "->".
It's like if the class is a " box " with inside some function, variable etc... the obj is a pointer to that box and you can see inside with the operator "->".
That box are defined here: 
`i2cEncoderLibV2 RGBEncoder[ENCODER_N] = { i2cEncoderLibV2(0x40),
                                          i2cEncoderLibV2(0x20), i2cEncoderLibV2(0x60), i2cEncoderLibV2(0x10),
                                          i2cEncoderLibV2(0x50), i2cEncoderLibV2(0x30), i2cEncoderLibV2(0x70),
                                          i2cEncoderLibV2(0x04), i2cEncoderLibV2(0x44),
                                        };`
Where ENCODER_N it's 9, so in that case i have defined 9 time the i2cEncoderLibV2 calss.

Gandrasg, 2020-12-21:
...that is clear. thanks
And can I do that out of the box? I want to load the counter data into an array.
```
if (obj->readStatus(i2cEncoderLibV2::RINC) && ledState1 == HIGH) 
        {
          P1_sliderValue[0] = obj->readCounterInt(); // 
```

Gandrasg, 2020-12-22:
I am slowly beginning to understand how it works. :)
After adding my function to the events the code above started working :) 
This is a great little structure! Love it.

```
    RGBEncoder[enc_cnt].onChange = encoder_rotated;
    RGBEncoder[enc_cnt].onChange = encoder_display; // it is not working properly yet
    RGBEncoder[enc_cnt].onChange = encoder_to_array; //ok
    RGBEncoder[enc_cnt].onButtonRelease = encoder_click;
    RGBEncoder[enc_cnt].onMinMax = encoder_thresholds;
    RGBEncoder[enc_cnt].onFadeProcess = encoder_fade;
```

Gandrasg, 2021-01-03:
Thanks!
Happy New Year!
Meanwhile...I’ve noticed on the encoders that as the number of IDs increases, I reach the maximum for more and more rotations.
Four encoders line up, the fourth causes much more rotation and triggers the same change as the first. Do you have any idea what could be causing this? There is only one SJ8 soldered in the queue and I turned off IPUP in the code.