DuPPadotnet/ArduinoDuPPaLib #25: Launch All state reactions(interrupt ) if a second encoder is not connected

fixus971, 2021-12-01:
Hi. There is a bug whereby if an encoder is missing(optional) and I call updateStatus().
In this case All interrupt routines are launched for
Encoder is released!
Encoder is pushed!       
Encoder is long pushed!  
Encoder is double pushed!
..
all this because this function does not check the error reported by the I2C bus
_stat = readEncoderByte(REG_ESTATUS);
receive -1 = 255

in lib i don't see a test function to check the connection.

I would add a "connected" boolean during setup or reset or begin

For the rest I am fine .. thanks
-Federico

fixus971, 2021-12-02:
Hi,

> What you mean "missing"?

I mean that the same program works with multiple hardware configurations: 2 or 4 encoders.
Where I have only 2 encoders mounted, the software of the absent ones triggers all the interrupts because the internal status read function does not return the value of the state but -1 as an error.. only that it is not managed and continuing the state 255 has all the event bits active.

Fattoresaimon, 2021-12-03:
Well in that case you can make the program that when it start check if the encoders are connected or no, and maybe save in some EEPROM if the configuration is 2 or 4. You can use the ID register for checking.
Looks more optimized than checking every time the errors.

fixus971, 2021-12-03:
Hi, I agree to have to do the initial configuration test.
I would need a bit of memory in the class.
I already use the ID register to hold the previous CVal value ..
It would be useful to also have this data in the class: last value read.
However, I am of the opinion that an if would be fine for the reception check after reading the status.
Put that an encoder gets damaged or comes off .. the CPU will get 100% clogged.
Thanks,
-Faith