DuPPadotnet/ArduinoDuPPaLib #32: I2C Encoder Mini - Interrupt does not go Low

xelawe, 2023-10-01:
I recently purchased a I2C Encoder Mini and attached it to an Arduino Duemilanove (  - -> GND,  + -> 5V, SDA -> A4, SCL -> A5, INT -> A3) and uploaded the example Basic_with_Callbacks.ino.

The INT Pin always stays on 5V. (Checked also wit oscilloscope)

If i just execute Encoder.updateStatus() in the loop() without "if (digitalRead(IntPin) == LOW) " all Callback-Routines are executed and output on the serial Monitor.

Am i missing something in the setup()?

Coding of example
  Encoder.reset();
  
  Encoder.begin(i2cEncoderMiniLib::WRAP_ENABLE
                | i2cEncoderMiniLib::DIRE_LEFT | i2cEncoderMiniLib::IPUP_DISABLE
                | i2cEncoderMiniLib::RMOD_X1 );

Fattoresaimon, 2023-10-01:
Hello,
Can you please check if the ID code is correct?
You can use the method Encoder.readIDCode()
Or run the example Change_Address

Check also the connections...
Thank you,
Simone

xelawe, 2023-10-01:
This is the output on the console, when i run Change_Address (Change from 0x20 to 0x21 and back):
```
**** I2C Encoder Address changer! ****
All the devices must have different address!
Avaiable commands:
  S: for searching the I2C encoder mini
  0xXX or XXX: Address of the target device in hex or decimal

Scanning....
I2C Encoder mini found at 32 ( 0x20 ) !!

All the devices must have different address!
Avaiable commands:
  S: for searching the I2C encoder mini
  0xXX or XXX: Address of the target device in hex or decimal

Insert the desired address in hex or decimal: 
Changing address from 32 ( 0x20 ) to 33 ( 0x21 )
Error in changing address!

All the devices must have different address!
Avaiable commands:
  S: for searching the I2C encoder mini
  0xXX or XXX: Address of the target device in hex or decimal

Scanning....
I2C Encoder mini found at 33 ( 0x21 ) !!

All the devices must have different address!
Avaiable commands:
  S: for searching the I2C encoder mini
  0xXX or XXX: Address of the target device in hex or decimal

Insert the desired address in hex or decimal: 
Changing address from 33 ( 0x21 ) to 32 ( 0x20 )
Error in changing address!

All the devices must have different address!
Avaiable commands:
  S: for searching the I2C encoder mini
  0xXX or XXX: Address of the target device in hex or decimal

Scanning....
I2C Encoder mini found at 32 ( 0x20 ) !!

All the devices must have different address!
Avaiable commands:
  S: for searching the I2C encoder mini
  0xXX or XXX: Address of the target device in hex or decimal
```
output of  Serial.println(Encoder.readIDCode()): 
`57`

Fattoresaimon, 2023-10-01:
Hello,
Looks like the I2C Encoder mini is working and the I2C connection is right.
Did you modified something from the setup?
The most important part  fort the INT configuration are the following:

pinMode(IntPin, INPUT);

and 

  // Definition of the events
  Encoder.onIncrement = encoder_increment;
  Encoder.onDecrement = encoder_decrement;
  Encoder.onMax = encoder_max;
  Encoder.onMin = encoder_min;
  Encoder.onButtonPush = encoder_push;
  Encoder.onButtonRelease = encoder_released;
  Encoder.onButtonDoublePush = encoder_double_push;
  Encoder.onButtonLongPush = encoder_long_push;

  /* Enable the I2C Encoder V2 interrupts according to the previous attached callback */
  Encoder.autoconfigInterrupt();

Also you need to enable the pull-up on the INT pin.

Thank you,
simone

xelawe, 2023-10-01:
No i didn't change the setup, i just used the example without modifying.

> Also you need to enable the pull-up on the INT pin.
I added an external 4k7 resistor

I added the Encoder.readIDCode() after the Encoder.reset, but it just returned 255. When i insert delay(500) before it returns 57.
 ```
 Encoder.reset();
  delay(500);
  Serial.println(Encoder.readIDCode());
```
Seems that there is a problem with timing on i2c. The Encoder Mini is the only device on the bus.

Fattoresaimon, 2023-10-01:
Now it works?
On the Reset function there is a 10ms delay...

xelawe, 2023-10-02:
No, it still doesn't work

Fattoresaimon, 2023-10-03:
Hello,

Ok can be a faulty board. You can try to flash again the board if you can.
Or send me a email i will send you a replacement,
Thank you,
Simone

xelawe, 2023-10-10:
Hello Simone,

i have three boards and now tested two of them and it is the same behaviour.
How can i reprogram/flash it? 
Additionally i found this issue: [https://github.com/Fattoresaimon/I2CEncoderMini/issues/5#issuecomment-765292563](url) Can this be the reason for the behaviour?

Thank you for your support so far!

BR Alex

Fattoresaimon, 2023-10-11:
Hello Alex,
The link doesn't work.

I have found the issue and is a mistake in the programming.
The INT pin is shared with also the RESET and UPDI function, by default the pin is configured as UPDI pin and not as GPIO.
Looks like the programmer didn't programmed the fuse bit of the Attiny so it didn't changed the function from UPDI to the GPIO.
For programming i use the Pickit 4, there are also the some opensource programmer like this:
https://daumemo.com/diy-updi-usb-programmer-which-can-be-made-with-cheap-hardware/

Pay attention because if you change the function from UPDI to the GPIO you can't reprogram the chip without a proper programmer.

xelawe, 2023-10-29:
Hello Simone,

i managed now to build the UPDI Programmer with an Arduino Nano. I successfully set the fuse with this command:
avrdude -C avrdude.conf -v -p attiny402 -c jtag2updi -P COM4 -Ufuse2:w:0x02:m -Ufuse5:w:0xc1:m -Ufuse6:w:0x04:m -Ufuse8:w:0x00:m -Ulock:w:0x00:m -Uflash:w:\Downloads\I2CEncoderMini.hex
(see also https://github.com/Fattoresaimon/I2CEncoderMini/issues/5#issuecomment-767573296)

Now the INT Pins goes LOW like it should!

THX!

BR Alex

Fattoresaimon, 2023-10-31:
Hello,
Great!
I have also solved the issue on my side!
Thank you!

closed 2023-10-31 by Fattoresaimon