DuPPadotnet/ArduinoDuPPaLib #35: Seperate callbacks for button-down and button-up instead only PUSHP?

vanderbreye, 2024-09-06:
The PUSHP event, which is supposed to trigger on button press in the library, is only fired when the button is released, rather than immediately when it is pressed. I need the button press event to trigger as soon as the button is physically pressed, without waiting for the release. With this function, i could implement a function for a long-press that doesn't trigger the short-click.

Fattoresaimon, 2024-09-11:
Hello,
This can be only of problem of interrupt calling order

```
if (_stat & PUSHR) {
    eventCaller (&onButtonRelease);
  }
  if (_stat & PUSHP) {
    eventCaller (&onButtonPush);
```
First i call the release and after the push.
Did you also enable the interrupt output of the push?

vanderbreye, 2024-09-11:
How to enable the interrupt  output? Could you provide an example? <3

Fattoresaimon, 2024-09-11:
There is an example here, i assume you are using the I2C encoder V2
https://github.com/Fattoresaimon/ArduinoDuPPaLib/blob/master/examples/I2CEncoderV2/README.md

```
Encoder.onIncrement = encoder_increment;
Encoder.onDecrement = encoder_decrement;
Encoder.onMax = encoder_max;
Encoder.onMin = encoder_min;
Encoder.onButtonPush = encoder_push;
Encoder.onButtonRelease = encoder_released;
Encoder.onButtonDoublePush = encoder_double_push;

Encoder.autoconfigInterrupt(); // Enables the I2C Encoder V2 interrupts according to the previously attached callback
```

Or manually with the command 
`void writeInterruptConfig(uint8_t interrupt)`